Sapd/HeadsetControl #375: Arctis nova pro wireless not detected

TheLaGmAn, 2024-10-08:
### Description

Im using W11, the DAC is connected and the compiled headsetcontrol.exe ver 3.0.0 is throwing:

d:\>headsetcontrol
No supported device found

anything im missing?

### Headset Name

Arctis nova pro wireless

### On which OS does the problem happen?

Windows

### Device information

<details>
 <summary>Detailed Device Information</summary>

Device Found
  VendorID: 0x1038
 ProductID: 0x12e5
  path: \\?\HID#VID_1038&PID_12E5&MI_03#9&36c2d62e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer: SteelSeries
  Product:      Arctis Nova Pro Wireless
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12e5
  path: \\?\HID#VID_1038&PID_12E5&MI_04&Col01#9&12eb986c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer: SteelSeries
  Product:      Arctis Nova Pro Wireless
  Interface:    4
  Usage-Page: 0xffc0 Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12e5
  path: \\?\HID#VID_1038&PID_12E5&MI_04&Col02#9&12eb986c&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer: SteelSeries
  Product:      Arctis Nova Pro Wireless
  Interface:    4
  Usage-Page: 0xff00 Usageid: 0x1

</details>

TheLaGmAn, 2024-10-08:
from what i see in /Sapd/HeadsetControl/tree/master/src/devices/steelseries_arctis_nova_pro_wireless.c
it has:
enum { ID_ARCTIS_NOVA_PRO_WIRELESS_BASE_STATION = 0x12e0 };

but my unit is
VendorID: 0x1038
ProductID: 0x12e5

are there different productids for the nova pro wireless?

Sapd, 2024-10-09:
> are there different productids for the nova pro wireless?

That is the case for almost all headsets, you can try adding it: https://github.com/Sapd/HeadsetControl/wiki/Development-1-%E2%80%90-Adding-a-device#check-if-a-similar-device-already-exists

nicola02nb, 2024-10-20:
@TheLaGmAn try with this build: 
[headsetcontrol.zip](https://github.com/user-attachments/files/17449599/headsetcontrol.zip)

nicola02nb, 2024-11-20:
@MasterYami Yep, only  adding the `ProductID `to `src/devices/steelseries_arctis_nova_pro_wireless.c` should do the job... 
If it works for you, you can go for PR....
I also got a a build with the same `ProductID` here:

> @TheLaGmAn try with this build: [headsetcontrol.zip](https://github.com/user-attachments/files/17449599/headsetcontrol.zip)

But I never got a reply from @TheLaGmAn ....

nicola02nb, 2025-09-17:
@Sapd can I suggest to close this issue?

> I added the product ID in steelseries_arctis_nova_pro_wireless.c and built and it seems

@TheLaGmAn is not answering and @MasterYami seems to have confirmed that by adding only the Id, like I have done in my PR, works flowlessly.

closed 2026-09-13 by github-actions[bot]