Sapd/HeadsetControl #381: HyperX Cloud III not detected

Miepee, 2024-11-30:
### Description

As the title says, the device isn't detected by headsetcontrol. I plugged it in via USB if that matters.

### Headset Name

HyperX Cloud III

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x03f0
 ProductID: 0x089d
  path: /dev/hidraw3
  serial_number: 000000000000
  Manufacturer: HP, Inc
  Product:      HyperX Cloud III
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x03f0
 ProductID: 0x089d
  path: /dev/hidraw3
  serial_number: 000000000000
  Manufacturer: HP, Inc
  Product:      HyperX Cloud III
  Interface:    3
  Usage-Page: 0x1c0 Usageid: 0x1

Device Found
  VendorID: 0x03f0
 ProductID: 0x089d
  path: /dev/hidraw3
  serial_number: 000000000000
  Manufacturer: HP, Inc
  Product:      HyperX Cloud III
  Interface:    3
  Usage-Page: 0x170f Usageid: 0x1

Device Found
  VendorID: 0x03f0
 ProductID: 0x089d
  path: /dev/hidraw3
  serial_number: 000000000000
  Manufacturer: HP, Inc
  Product:      HyperX Cloud III
  Interface:    3
  Usage-Page: 0xffc0 Usageid: 0x1

 ```

</details>

Miepee, 2024-11-30:
I tried basic print debugging and for some reason, the headset is never searched for?
Its vendor id is 0x03f0 = 1008
Printing all the vendor ids that get searched for in `find_device`, I see `9610` four times and `1133` once.

nicola02nb, 2024-12-03:
A good way to find your device ID is to run:
```
headsetcontrol --dev -- --list
```
Then find your device in the list reported.
Example for Steelseries Arctics Nova 7:
```
Device Found
  VendorID: 0x1038
 ProductID: 0x2202
  path: 
  serial_number:
  Manufacturer: SteelSeries
  Product:      Arctis Nova 7
  Interface:    3
  Usage-Page: 
```
Then probably you should add your `ProductId` to the supported `PRODUCT_IDS[]` at:
https://github.com/Sapd/HeadsetControl/blob/c96c3d9ce1893637c31c60c25d3403ec84083bc9/src/devices/hyperx_cloud_3.c#L8-L13

Miepee, 2024-12-03:
> A good way to find your device ID is to run:

See the output in the first message under "detailed device information".
You can see there that the vendor ID is `0x03f0` and the product ID is `0x089d` - both values already exist in the snippet you linked!

Hence why I tried to do the print debugging (my second message) and found out that it doesn't even ever try to search for the vendor ID.

nicola02nb, 2024-12-04:
> See the output in the first message under "detailed device information".

Oh, didn't see that, sorry...

nicola02nb, 2024-12-04:
I might have found the problem.
I should be there:
https://github.com/Sapd/HeadsetControl/blob/c96c3d9ce1893637c31c60c25d3403ec84083bc9/src/main.c#L61-L63
Because it stops checking the rest of the devices... probably you have an already compatible device connected, and when it finds it, it stop searching for other compatible devices(Like your hyperx cloud 3).

Unplugging the device foud before your hyperx cloud 3 should work.

nicola02nb, 2024-12-04:
It actually a problem of the entire software which was designed to support one device at a time...

Miepee, 2024-12-04:
> Unplugging the device foud before your hyperx cloud 3 should work.

I swear I tried it earlier and it didn't work. But now with the other device disconnected and several days/reboots later, it does work.

I wonder if maybe something internally caches something, when i have both plugged in, then plug out the other supported one and run headsetcontrol again. Not sure
Either way, thanks for finding that!

nicola02nb, 2024-12-04:
Sond strange OS chaching usb devices lol.

Tho, no problem

Sapd, 2024-12-04:
> It actually a problem of the entire software which was designed to support one device at a time...

Yeah there is some work already in fixing that. For example the JSON/YAML output functionality already fully supports multiple devices. However main problem is still the question of specifying which to use for sending commands. With multiple devices a user would need to somehow specify which device, which I did not think about yet of how to do that in a user friendly way.

> I wonder if maybe something internally caches something, when i have both plugged in

Should in theory not, as it simply uses the HID calls of the kernel to iterate (which in turn uses USB iteration). Maybe some weird kernel bug or edge case.

Miepee, 2024-12-04:
It wouldn't surprise me if I had done a kernel upgrade that day and forgot to reboot.

nicola02nb, 2024-12-06:
> However main problem is still the question of specifying which to use for sending commands

I got an idea there #382

tunsssss, 2025-01-03:
same issue here, on windows, i don't have any other compatible device other than the cloud 3, same vendor and product id as yours

closed 2025-12-31 by github-actions[bot]