Sapd/HeadsetControl #385: led lights control, possibiliti to set colors based on HEX or R G B values

y0ur5h4d0w, 2024-12-26:
### Description

hi! i didn't se the possibility to actually control which color is displayed by the headsets, i have a void elite and i can't find any software that allows me to control the actual color of the LED. neither openRGB or CKB-NEXT can send data to my headset to set the LED to for example 255 0 0 for the red color on the zones, it would be cool if it's possible to map out the 3 zones (2 lateral logos and 1 for the microphone activated or muted) and let us choose the color, i don't know if the mic activation led is actually RGB because i windows with iCUE it wasn't possible to choose the actual color. 

my idea was to go something like:

headsetcontrol --zone-logos 255 0 0 to set the lateral logos to red
headsetcontrol --zone-microphone 1 to turn on the microphone red led

hope it's doable! or if there's a different way to control the leds let me know :)

bgstack15, 2024-12-30:
If we could use software to control the individual mic led, it would facilitate turning off the "rgb" lights on the headphone cups for lower battery usage, while still showing a red light when the mic is muted! (The goal of https://github.com/Sapd/HeadsetControl/issues/57#issuecomment-614224798)

DogancanYr, 2025-02-18:
Are there any improvements?