Sapd/HeadsetControl #390: Add support for chatmix on SteelSeries Arctis Nova Pro Wireless

BurritoSpray, 2025-01-09:
### Description

It would be nice to have the chatmix feature working for the Arctis Nova Pro Wireless device.
I will probably checkout the wiki and poke arround so I can help implement it in my free time.

### Headset Name

SteelSeries Arctis Nova Pro Wireless

rhinotastic, 2025-03-24:
This may be of some help.
https://github.com/dfanara/Arctis-on-Linux/blob/master/README.md
and this other project
https://github.com/elegos/Linux-Arctis-Manager

BurritoSpray, 2025-03-25:
I found a working solution, [here](https://github.com/Dymstro/nova-chatmix-linux) and [forked](https://github.com/BurritoSpray/nova-chatmix-linux) it to improve a few things to my liking. From his research it should be fairly easy to implement the controls to control the chatmix mode and icon in the display. 
Ive been using it for two months now and it works great.
Other than that the mix side is all software so there is not much we can do about it.

rhinotastic, 2025-05-24:
> I found a working solution, [here](https://github.com/Dymstro/nova-chatmix-linux) and [forked](https://github.com/BurritoSpray/nova-chatmix-linux) it to improve a few things to my liking. From his [...]

Is this still working fine for you? i am very rusty with linux as i've only dabbled on and off over the years but i'd like to be able to us the chatmix at the least on my nova pro headset.
Did you use this solution along with headset control? Or on it's own without headset control?

rhinotastic, 2025-11-21:
@Anden Hey sorry for the ping but since you did the initial implementation of the nova pro wireless, can you possibly look at the chatmix dial functionality? there's some other projects that add the support but it would be good to have it in this one. I've not knowledge of c/c++ so i wouldn't know where to start to add how others have done it into what you already made.