Sapd/HeadsetControl #392: New HeadsetControl feature: Control parametric equalizer (with implementation for Arctis Nova 5)

ne0phyte, 2025-01-23:
### Description

Hi,

I added a new capability for parametric equalizers to HeadsetControl in my fork here:
https://github.com/ne0phyte/HeadsetControl/commit/c26bce3c013916ef06f0a8d70dfe1d91037ff5fb

I did not want to break the existing basic equalizer implementation so I added it independently as a capability, ParametricEqualizerInfo, struct parametric_equalizer_settings, struct parametric_equalizer_band and EqualizerFilterType.

I added YAML and JSON outputs, help texts, output of capabilities and the five filter types the Arctis Nova 5 supports for now. More filter types can be added if other headsets feature different filters and the supported filters are set by each devices implementation.

Bands are represented by the typical parameters of parametric equalizers: frequency, gain, q-factor (band width) and filter type.

This is the help text (which also gives an example of the format I chose to specify the bands):

```
Parametric Equalizer:
  -q, --parametric-equalizer STRING     Set equalizer bands (bands separated by semicolon)
      Band format:                      FREQUENCY,GAIN,Q_FACTOR,FILTER_TYPE
      Availabe filter types:            lowshelf, lowpass, peaking, highpass, 

      Example:                          -q '300,3.5,1.5,peaking;14000,-2,1.414,highshelf'
                                        Sets a 300Hz +3.5dB Q1.5 peaking filter
                                        and a 14kHz -2dB Q1.414 highshelf filter

```

I would like to ask whether my solution is acceptable before creating a merge request. Is there anything I should change?
And is adding --parametric-equalizer next to --equalizer okay?

Would appreciate some feedback

### Headset Name

Arctis Nova 5

Sapd, 2025-02-01:
Feel free to open up a PR. Looks fine on first look to me. 
Only one remark: Best would be to remove the short parameter `-q` and just keep the long one `--parametric-equalizer`. As the amount of letters are limited I would just use them for the most used features or flags for the program itself.

ne0phyte, 2025-02-01:
Thanks. I removed the -q flag and opened a PR.

closed 2025-02-01 by ne0phyte