Sapd/HeadsetControl #418: undefined reference to `hid_get_input_report'

sirshurf, 2025-08-14:
### Description

While running make, I am getting this error.
I tried installing hid using pip but it does not help.

[ 97%] Linking C executable headsetcontrol
/usr/bin/ld: CMakeFiles/headsetcontrol.dir/src/devices/audeze_maxwell.c.o: in function `send_get_input_report':
audeze_maxwell.c:(.text+0x245): undefined reference to `hid_get_input_report'
/usr/bin/ld: audeze_maxwell.c:(.text+0x26e): undefined reference to `hid_get_input_report'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/headsetcontrol.dir/build.make:565: headsetcontrol] Error 1
make[1]: *** [CMakeFiles/Makefile2:337: CMakeFiles/headsetcontrol.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

Linux Ubuntu 20.

### On which OS does the problem happen?

Linux

sirshurf, 2025-08-15:
Yep

libhidapi-dev is already the newest version (0.9.0+dfsg-1).

ModoloDev, 2025-08-15:
@sirshurf, idk if that is the problem, but audeze_maxwell.c is including the hidapi header with `#include "hidapi.h"` instead of `#include <hidapi.h>`
I've fixed it in a fork, could you test it, please?
https://github.com/ModoloDev/HeadsetControl/tree/bb9625034d340c2849b9a48fbe543f30ffd3f18d

sirshurf, 2025-08-15:
Nope

Same, I used your link
 git clone https://github.com/ModoloDev/HeadsetControl.git && cd HeadsetControl
And this is what I got.

/home/sirshurf/HeadsetControl/src/devices/audeze_maxwell.c: In function ‘send_get_input_report’:
/home/sirshurf/HeadsetControl/src/devices/audeze_maxwell.c:102:42: warning: implicit declaration of function ‘hid_get_input_report’; did you mean ‘send_get_input_report’? [-Wimplicit-function-declaration]
  102 |             res                        = hid_get_input_report(device_handle, tempBuff, MSG_SIZE);
      |                                          ^~~~~~~~~~~~~~~~~~~~
      |                                          send_get_input_report

[ 97%] Linking C executable headsetcontrol
/usr/bin/ld: CMakeFiles/headsetcontrol.dir/src/devices/audeze_maxwell.c.o: in function `send_get_input_report':
audeze_maxwell.c:(.text+0x245): undefined reference to `hid_get_input_report'
/usr/bin/ld: audeze_maxwell.c:(.text+0x26e): undefined reference to `hid_get_input_report'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/headsetcontrol.dir/build.make:565: headsetcontrol] Error 1
make[1]: *** [CMakeFiles/Makefile2:337: CMakeFiles/headsetcontrol.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

ModoloDev, 2025-08-15:
> Yep
> 
> libhidapi-dev is already the newest version (0.9.0+dfsg-1).

Oh, I see what’s happening.
The latest versions isn't 0.9.0, it's [0.15.0](https://github.com/libusb/hidapi/releases/tag/hidapi-0.15.0)
`hid_get_input_report` was implemented in [0.10.0](https://github.com/libusb/hidapi/releases/tag/hidapi-0.10.0)